mxtommy/SigkSens #36: If FS can't be opened, handle it better

ba58smith, 2018-08-26:
Now, the program just runs merrily along. But w/o the FS working, the only configuration that can be saved is the wifi config, because ESP8266 stores that info in a specific place in memory. Things to consider for better handling:
-  A more complete error message in Serial Monitor. Possibly mention that the MCU's SPIFF setting may be "Off", and suggest checking that. (That was my problem.)
- Make the initial config screen display something at the end - "Configuration failed: file system could not be accessed. Check SPIFFS setting on your MCU."

mxtommy, 2018-09-19:
In 37b586635c2940d85d79524b3455a67d5864581d I changed it so it will print a more detailed error message and stop. Unfortunately the wifi config screen can't be modified easily as it's part of the library. Still, should be more clear :)

closed 2018-09-19 by mxtommy

ba58smith, 2018-09-20:
<two thumbs up!>