mxtommy/SigkSens #40: How to add support for other boards

joabakk, 2018-09-16:
Like ESP32, I understand the libraries are different, so some kind of logic is required. Has anyone tried this?

ba58smith, 2018-09-17:
@joabakk I'm not sure I understand your question. Do you mean has anyone tried to use SigkSens with an ESP32? I think @mxtommy has done some work, but not all the way to getting it working, or maybe not in all situations.

I spent some time looking for a recent article on migrating from the 8266 to the 32, and didn't find anything that was excellent (although I did find a lot of info). Here's one article that has a simple Arduino Sketch that works for both 8266 and 32, and it highlights the differences. https://www.instructables.com/id/ESP8266-and-ESP32-With-WiFiManager/ That may be a good place to start. But unless you really need some feature of the ESP32, why not just use theESP8266? They're readily available and cheap, and they work now, with no experimentation necessary.

BTW, if you haven't looked at the Wiki pages yet, do that - that's the most current documentation.

joabakk, 2018-09-17:
OK thanks, I bought a few EPS32's with screens and all, and thought it would be a matter of small tweaks to get it working with SigKsens. Classic mistake. But I can probably make use for them anyway

mairas, 2018-09-17:
It _may_ be a matter of small tweaks. I think @mxtommy tried to get ESP32 working some time ago but didn't succeed then. AFAIK, all the dependent libraries nowadays support ESP32 as well, so the changes shouldn't be drastic. You're right, though, that some libraries use e.g. different include names for ESP32. For these, #ifdefs would need to be added.

If I may, I'll reopen the issue because I'd say ESP32 support is a valid request for the project. Brian is right though, that for the time being, it's better just to use ESP8266, unless you specifically want to hack with the codebase.

mxtommy, 2018-09-18:
It should be possible to support the ESP32 without TOO much work :) I tried it with a test sketch, and the big differences are in Wifi, and mDNS/discovery libraries weren't there/compatible. May have changed in the months since I played with it though.

 I'll say the little I played with the ESP32 I found it is still a bit immature compared to the ESP8266. (I'm talking about Arduino Core) That's not to say it's not 99% there, just that there are still some bugs and that googling for help returns far fewer results :) As an example, I was not able to get i2c connection to an MPU9255 working with the ESP32. Googling showed some other people having the same difficulties, and some potential work arounds, but in the end I couldn't get it working. (confirmed problem was code though with a scope/protocol decoder. Right info was on the the i2c bus but code failed :( 

Still, a valid goal as ESP32 has more of that sweet sweet ram :D

joabakk, 2018-09-20:
Good, thanks for re-opening. Perhaps we should branch out and see what is missing. I had the same experience with the MPU9255. But the screen would be a welcome addition to a holding tank sensor designed to cater for an analog gauge. Stick the M5Stack in and get both a gauge and SK data! And as @sbender9 suggested, a warning to only eat salad if you flush twice!

ba58smith, 2018-09-20:
I would also like a holding tank sensor, but I assumed there would be a way
to display it in SK, making support for a screen in SigkSens unnecessary.
Is that not the case?

Butch

joabakk, 2018-09-21:
That is the case, yes. The display I can add on the side. I could even use the M5Stack as a pure client for local display of that sensor.

ba58smith, 2018-09-21:
@joabakk - what sensor are you thinking of using for your holding tank? I have a thick plastic tank held in place with an angle iron framework, and my first attempt at using an MPR121 capacitive sensor (with strips of copper tape on the outside of the tank) failed, I think because I had to run long wires from the Arduino to the copper strips, and they had to go along the angle iron framework. I want to try it again, this time with an ESP8266 that's right next to the copper strips, with very short wires, not close to the angle iron at all. But if there's a better way to do this - I'd love to know it! (My Tank Monitor works, but when the red "Full" light comes on, the tank is FULL!!! I want more detail for the top 1/3 of the tank.)

joabakk, 2018-09-24:
I have a Gobius ultrasonic system. Works well and does not require leak testing...

mxtommy, 2018-12-20:
FYI ESP32 support is in testing :) (there's a branch for esp32 support, you can check PR #60 :) )

Going to close this issue as honestly I don't think there will be support for anything other and ESP8266/ESP32 for now. :)

closed 2018-12-20 by mxtommy

ba58smith, 2019-05-06:
JK,

The trick with getting anything on the N2K bus. Signal K, with some
specific hardware, can do that, but I haven't gotten that far yet. I know
you'll need to buy a "converter" that will take your Signal K data and add
it to the N2K bus - for some reason, you can't do that yourself, or if you
could, it's really hard.

I'm still just working on getting my data to display on the Signal K
Instrument Panel. In fact, today was the first time I did that while we
were underway! I put four temp sensors on the main engine (coolant temp,
exhaust elbow temp, 12V alternator temp, and 24V alternator temp),
connected them all to an ESP8266 (Wemos D1 Mini), running SigkSens (
https://github.com/mxtommy/SigkSens), and sent the data to my Signal K Node
Server running on my Raspberry Pi!  SigkSens is the project I've done
documentation for (the entire Wiki), and have been focusing all of my
hardware efforts on. It's not very mature, unfortunately. @mxtommy/SigkSens
<reply@reply.github.com>
,
who created it, has gotten it to a point that it's doing most of what he
wants it to do for his boat, but hasn't done much else. And a few months
ago, he started a whole new branch (the Development branch) to completely
redo the way it defines and interfaces with sensors. It was maybe not
especially well designed, and he stopped mid-project, and that's where it
sits. So, if you DO do anything with it, use the Master branch. You can try
other microcontrollers, but the Wemos D1 Mini (NOT the "Mini Lite"
version!) works well with it, and you can buy them for under $10 each.

Yesterday, I built a tachometer for my main engine, designed to use the
tach wire from the alternator as an input to an opto-isolator, that then
inputs into the Wemos. It works perfectly on the bench, using an old
Arduino Uno to provide the pulses to be measured. But as soon as I connect
it to the actual alternator, the output from SigkSens is about 10% higher
than it should be. I spent all day today working on it, and got nowhere.
Driving me crazy. Now that I have it built, I may add a Hall Effect Sensor
(senses a magnet passing close by) to it, and see if that works better. I
gave up on using the IR LED / phototransistor - that's what an
opto-isolator is, and they're cheap.

Do you know about the website Slack? There's an active Signal K developer's
space there that I've used several times to get quick answers. That would
be a great place to ask exactly how you get Signal K data into N2K.
signalk-dev.slack.com. All of the chatter about SigkSens happens in the
#sensors Channel, or in private discussions between individuals.

I do know that most analog gauges - especially temp and pressure gauges -
are essentially voltmeters, and the sending units are just variable
resistors. Please tell me about the gauges you're planning to make - I'd
like to have one for oil pressure.

We've been in the Virgin Islands since March 28. We've been to a lot of the
places here, but most of them for only a day or so. We've spent a lot of
time in a few bays that are pretty protected from the damned wind that
never seems to blow less than 15 knots. We were really looking forward to
getting here, but honestly, it's a bit of a disappointment. I think it
would be great for a sailboater, but for us - that damned wind!

Are you going to be in Indy for Thanksgiving this year? We will be - and
probably in FL for a few days around then. (I have to renew my FL driver's
license, and this time, it has to be in-person.)

BS

Brian "Butch" Smith
M/V Smartini
www.smartini.life

On Mon, May 6, 2019 at 2:45 PM Joel Kozikowski <joel.kozikowski@gmail.com>
wrote:

> BS -
>
> When I was last visiting you, you said you were writing some documentation
> for some signal K project? Which specific project was that?
>
> I have been doing a little more research on the way that analog gauges
> work. You may already be aware of this, but most analog gauges, be them
> temperature or oil pressure are really nothing more than a voltmeter.
> There is a signal line, and the difference in temperature is encoded as a
> difference in voltage.
>
> Anyway - I plan on making drop in replacements for my generator gauges
> that display the values in digital, but ALSO broadcast the values over wifi
> so I can pick them up and put them on my N2K bus to be displayed on my MFD.
>
> I don’t want to reinvent the wheel however, so I wanted to see what work
> has being done on the project you are already working on it.
>
> -jk
> (sent from my iPhone)