mxtommy/SigkSens #49: Output 1-Wire address(es) to Serial Monitor as they're discovered

ba58smith, 2018-09-25:
When you scan the 1-wire bus at startup, output the address of any discovered sensors in the Serial Monitor. As it is, you have to run /getSensorInfo to know what the address is, so you can then run /setSensorAttr. Trying to save a step.

mxtommy, 2018-09-26:
hmmm, not so simple after all :)

On first bootup (before it's ever seen the temp sensor) it will actually print it already. If it's "known" already (already in config) it doesn't print. This is because we scan for new sensors every 30 seconds. (in case of bad connections or so). on subsequent bootups though you should see the address when it prints the config :)

Will leave it as is for now :)

closed 2018-09-26 by mxtommy

ba58smith, 2018-09-26:
I hadn't seen it in the config output on each reboot. That's just as good - saves a step. Thanks for pointing it out.