mxtommy/SigkSens #53: Change digitalIn state to be "true" when "HIGH"

ba58smith, 2018-10-01:
Change LOW to HIGH in the method below, in digitalin.cpp.

```
  if (si != nullptr) {
    //current state
    if (strcmp(si->signalKPath[0].c_str(), "") != 0) {    
      if (digitalRead(digitalPins[index]) == LOW) {  //Change LOW to HIGH, so voltage on the pin = "true"
        si->valueJson[0] = "true";
      } else {
        si->valueJson[0] = "false";
      }
      si->isUpdated = true;
```

mxtommy, 2018-10-02:
Fixed in 3b407400d25d294a6f0dd81537b7a169d034f68d It's now defaulting to HIGH, but can be changed easily in config.h

closed 2018-10-02 by mxtommy

ba58smith, 2018-10-02:
Does this code, also in digitalin.cpp, also need to change?
```
  if (si != nullptr) {
    //current state
    if (strcmp(si->signalKPath[0].c_str(), "") != 0) {    
      if (digitalRead(digitalPins[index]) == LOW) {
        si->valueJson[0] = "true";
      } else {
        si->valueJson[0] = "false";
```