mxtommy/SigkSens #59: Davis wind sensor / mast head sensor

dmdelorme, 2018-12-06:
I am rewriting my code for a masthead unit that sends back wind speed and direction and receives message to control the anchor lite over the air.
The current unit is nrf24 based. With code on the rpi (openplotter) to write to local port and read pin state to send message to the MHU to control the anchor light via Mosfet.
Power is supplied by wire up mast. The current mast hardware is a Nano with a NRF24l radio Davis 6410.
I am doing a new version with a Adafruit HUZZAH32 – ESP32 Feather Board. 
[image]
The current system uses 5 volt for the digital and ADC.
I am wondering if i could feed the Davis circuit 3.3 Volts and avoid level shifting and  a voltage divider  on the ADC pin.. ???
The only issue i see is voltage drop on the 6410 but the wire connecting it will be short not 40+ feet.
I have yet to read though your code but I would like to add this as a device/sensor.
.The new code i use sends NMEA
` $ARMWV,163.50,R,0.00,N,A*2f`
and SignalK creates 
```
"OPkplex.AR.environment.wind.angleApparent"
"OPkplex.AR.environment.wind.speedApparent" 
```
``` 
I tried the other way around to send SignalK but had issues with openplotter/SignalK generating a MWV sentence.
The new design should be much smaller. and fit in a case under the masthead light. A battery could be added as a back up power source.

mxtommy, 2018-12-20:
I must admit I'm not familiar with the that windvane or how it outputs direction/speed. (I guess the reed switch is a pulse ouput for the wind speed and the yellow/green wires together indicate angle?)

Regarding running it off of 3.3V instead of 5, I don't think the reed switch would impact anything, For the direction I would imagine there is math to turn the 2 voltage levels to an angle, and that math may change? I don't know the math though. Do you? :)

As for sending SignalK and having issues generating a MWV sentence, I think that's our of scope of this project. That said if you come join us on SignalK's slack [here](http://slack-invite.signalk.org/) there are plently of people there who could help getting that going :) (even with you existing setup :) )

dmdelorme, 2018-12-21:
I worked just fine with 3.3v  supplied from the esp32 feather not much for voltage drop and it read the full range @ 12bit.
I did re profile the voltage before mapping Deg. My first pvc board is in the mail?
It acts as a base for my Anchor light at the mast head i have the light controlled via mosfet on esp32 and hard switch tied to a GPIO pin on RPI.