mxtommy/SigkSens #61: Add support for a Tank Level Sensor (based on digitalIn pins)

ba58smith, 2018-12-20:


ba58smith, 2018-12-20:
Based on some Slack chat: create a Tank Level Sensor that outputs a decimal value (0.0 to 1.0), based on the state of X number of digital input pins. For example:
pin 0 = "Empty", outputs 0.0
pin 1 = "Low", outputs 0.25 (or whatever the user configures it to output)
pin 2 = "Half", outputs 0.5 (or whatever is configured)
pin 3 = "Full", outputs 0.95 (or whatever is configued)

In that order, output the value related to highest pin number whose "state" is "HIGH".

(Or something like that!)

mxtommy, 2018-12-20:
Simple enough :) I have 2 weeks off starting next week will try and get it done :D

ba58smith, 2018-12-20:
That would be great! Then, I might actually put one of these things in place on my boat, finally. (I'm great at writing about it, not so great at execution.)

mxtommy, 2019-05-22:
Was done in dev version :) might still need some tweaks let me know :)

closed 2019-05-22 by mxtommy