mxtommy/SigkSens #62: SensorInfo

mxtommy, 2018-12-20:
The following is strictly brainstorming for the moment :) 

So, SensorInfo while working great, seems potentially overly complicated/verbose? Each sensor has 80+ lines of code to set it up. and there are ifdefs and includes in many places for it :) Need to setup SensorType/fromJson/etc as well.

I originally came up with the SensorList concept because oneWire was the first sensor I did, and it had to support many temp sensors each with an address that you didn't know about beforehand. All the others sensor types though are more or less "simple". There's usually only one instance of them. Might make sense to keep complexity inside those sensors code.

If we look at the requirements for what it does, I come up with the following:

- Allow saving the config of the sensor attributes (toJson)
- Allow loading the config of the sensor attributes (fromJson)
- Allow the http server a method for getting/setting an arbitrary list of attributes of the sensor. (path/scale/offset/value/etc)
- Allow the sensor to send deltas.

I think we could implement it in a cleaner way than SensorInfo/SensorStorage?

- For Loading/saving config of sensor, I'm thinking it would be worth it to have a better "config storage" solution for everything. Ie a function to save "key ->value" that you can call from anywhere, anytime... There's likely some existing library that does this. :)
- For http getting/setting. Sensor on setup() could register what attributes it has with http and register a callback to get current values/set new ones? I'm not super good with callbacks so let me know if I'm crazy or there's another easier way :)
- Sending deltas is simple, just a function in signalk.cpp that accepts path->value and sends it rather than iterating through SensorStorage. When a sensor wants to send a delta use that function instead of setting isUpdated = true. 

I'm thinking it's a bunch or work, but would simplify adding new sensors, and make code less complicated I think :) I'm open to keeping SensorInfo/SensorStorage though it's a better solution.

ba58smith, 2018-12-20:
Are you talking about replacing (for example) SHT30SensorInfo::toJson(), and OneWireSensorInfo::toJson(), and ADSSensorInfo::toJson(), etc. with a more abstract approach like writeConfigToJson(sensorType, attribute, setting), that can be used for all existing and future sensor types? If so - that would seem to simplify adding new sensor types. But, as my Uncle Don would ask in this situation, is the juice worth the squeeze? I mean, are there that many more sensor types that you'll want to add in the future to make it worth the effort of simplifying the process? I can't say - I've never added a sensor type, nor am I likely to in the future. It would seem to be a fairly complex conversion process, prone to error, and requiring a lot of testing.

mxtommy, 2018-12-26:
I think it may be worth it for at least making the code more readable :) Played a bit with making a key/value store that saves to flash. Anywhere in the code you could do a simple Get or set for a config option. Will give it a bit more thought for the rest (checkout configStore.cpp inthe configKV branch if interested :) )

mxtommy, 2019-01-01:
Ok, did a bunch of work in #development branch. Would appreciate feedback (Positive or negative :D ) 

( Not ready to merge to master yet. All sensors but systemHz are not working )

- First LedBlinker has been moved to it's own file :) 
- Coded up a Key->Value store in configStore.cpp. It saves KV pairs to flash in json format. I'm sure it could be made more "pretty" but it's working :)
- Re-wrote signalk.cpp to use a c++ map instead of iterating through SensorInfo. You call signalk.addValue(path, value) and it gets sent out :)
- Refactored systemHz.cpp to use the configStore/new method to send deltas. Seems much more readable (and shorter/simpler!) to me. 

Still to do is the HTTP stuff. I'm thinking instead of all the different paths, just exposing the key/value config. With a simple webpage app would make it simple to configure stuff :) 

Let me know what you think! :)

mairas, 2019-01-01:
Replying to Tommy's initial message only: this is pretty much what I have been thinking about as well while walking the dog! :-D

Would it be better if the config storage was generic Json instead of key-value pairs? Let's say you're implementing a compass sensor and want to store a deviation calibration table, that'd probably be quite a bit nicer with native Json than key-value pairs. In a similar fashion, instead of letting the individual sensor component to register HTTP configuration endpoints, the configuration could happen with Json snippets.

What else? Oh yeah, naming: we call the whole physical device with the individual sensing hardware  "sensors", but we call those bits of sensing hardware "sensors" as well. Both can be justified, but the result is that we have sensors connected to sensors. My brain kinda melts down when I try to think of a better naming scheme, though. Could the physical sensors be called "components"? After all, they might not be even sensors but actuators or maybe even displays?

One minor detail about the current implementation that I've been planning to fix for a long time: it's generally preferable to save whatever data you have in native format (C++ variables or class objects) and then serialize when needed. That is, when the hardware component value is read, the value should be saved to native variables. Creating the Json output string should happen only when that output is needed.

ba58smith, 2019-01-01:
I can't comment on the code, but I like the idea of having a simple webpage to configure sensors. /setSensorAttr works, but isn't the easiest thing to use.

ba58smith, 2019-01-01:
RE: naming. When I started to write documentation, I also got "sensors" and "sensors" confused. I decided on "MCU" for the physical device (Wemos D1 mini, ESP32 Wrover Module, etc.) and continued to call anything that brought in a reading of some kind a "sensor". "Sensor" works for all of the ones that are already defined, except DigitalOut, and if that remains the only exception, it's not too confusing to think of that as another "sensor" with some different behavior. But it could be called an "Output", I suppose, especially if there are other ones besides DigitalOut at some point in the future. I wouldn't think that any kind of display would be configured at all like the rest of the outputs that might end up existing. The little LED display I've been playing with certainly isn't used at all like DigitalOut.

mxtommy, 2019-01-01:
Re: naming If there's no objection, I'll rename sensors directory as "components". I think it makes sense :)

closed 2019-05-22 by mxtommy