mxtommy/SigkSens #63: Analogin A0

dmdelorme, 2018-12-23:
On a esp32 i believe that A0 can not be used when Wifi is active.. 
I had this issue with a esp32 feather.
so i set the analog pin to ` WDIR_PIN = 34;// A2/GP34 ADC#1 so it will work with wifi`

dmdelorme, 2018-12-23:
[https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html](url)

dmdelorme, 2018-12-23:
I also had to fix the issue with the adc not being perfectly linear. I used this little bit of  code so my wind vane would point in the correct direction..   
```
`double ReadVoltage(byte pin) {
  double reading = analogRead(pin); // Reference voltage is 3v3 so maximum reading is 3v3 = 4095 in range 0 to 4095
  if (reading < 1 || reading > 4095) return 0;
  // return  * pow(reading,3) + 0.000000016557283 * pow(reading,2) + 0.000854596860691 * reading + 0.065440348345433;
  return (-0.000000000000016 * pow(reading, 4) + 0.000000000118171 * pow(reading, 3) - 0.000000301211691 * pow(reading, 2) + 0.001109019271794 * reading + 0.034143524634089) * 1241.2121;
} // Added an improved polynomial, use either, comment out as required`

some one else's code but it corrected the issue of the ADC and the Wind vane positions matching ie 180 deg now is half of the adc reading.  the ADC curve is a little flat at both ends. 
```

dmdelorme, 2018-12-23:
After some reading I might found the issue with the calibration.. [https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html#adc-calibration](url).
My chip does not have the efuse set hence my calibration is off.
`espefuse.py --port /dev/ttyUSB0 adc_info
espefuse.py v2.5.1
Connecting.....
ADC VRef calibration: None (1100mV nominal)`

ba58smith, 2018-12-23:
Nice digging, @dmdelorme!

Haimana, 2020-02-17:
Hello, is it possible to have more analog sensors with ESP32 but without ADS1115? 
Thanks,
Seb

ba58smith, 2020-02-17:
@Haimana - It depends on which version of the ESP32, but the base chip has 18 Analog In channels. My Heltec ESP32 with built-in OLED screen has all 18 analog channels exposed as pins, but some exspose fewer. (Unlike the ESP8266 which has only ONE Analog In pin.) So yes, physically, you have plenty of analog pins on an ESP32.
However, since SigkSens was originally designed for the ESP8266, which has only ONE Analog pin, all of the code assumes only one AnalogIn sensor, and I think it would be a lot of work to modify it to accept multiple Analog inputs.
SensESP, a similar project to SigkSens, was also written assuming only one Analog pin, however, it would be quite simple to modify it to accept multiples.
HOWEVER - neither SigkSens nor SensESP has been used with any ESP32 boards for more than just getting them to compile, so really, you're better off not using an ESP32. You're likely to have a lot of problems getting it to work, and no one is actively trying to make it work with ESP32.
Get the Wemos D1 Mini and as many ADS1115's as you need.

Haimana, 2020-02-17:
I understand, thank you for your detailed answer.  So back to ESP8266 :)

closed 2020-02-20 by mxtommy