mxtommy/SigkSens #65: limit the rate of messages

marijnacht, 2018-12-24:
It would be great to be able to limit the number of messages/sec going to the signalk server. Now I have a BMP280 sensor and barometer pressure and also temperature does not change very quickly. 1 message/minute would be fast enough instead of 1/sec now. Is there an easy way to achieve this?

ba58smith, 2018-12-24:
Ideally, it would be specific to each sensor, but that may not be practical. e.g., I don't care to see an updated outside temperature every second, but I DO want to see an updated turbocharger temp every second. The code to determine each sensor's update frequency might take as long to execute as the code to request the sensor's value, in which case, might as well just update it. If it's not changing, your display won't change no matter how often you update it.

mxtommy, 2018-12-26:
As Brian says there's no harm in sending the values more often. I'd even argue it probably helps keep the wifi stable by keeping a steady stream of data going. That said there shouldn't be much harm in changing the interval to a little slower.

Should be simple enough to change it.

https://github.com/mxtommy/SigkSens/blob/master/SigkSens/src/sensors/bmp280/bmp280.cpp#L96

On that line, change "SLOW_LOOP_DELAY" to whatever interval you want (in ms, so 1 second = 1000)

Note, the code on lines 106/107 of bmp280.cpp help smooth out the raw reading via a simple exponential filter. That might need tweaking if you put too long an interval or the values wont change quickly enough. :) Feel free to reopen issue if you have any other questions/comments :)

closed 2018-12-26 by mxtommy