mxtommy/SigkSens #68: LED_BUILTIN isn't defined for ESP32.

ba58smith, 2018-12-27:
From my summary of the getting SigkSens to work with ESP32:

5. SigkSens has (I think) three references to LED_BUILTIN, which isn't defined for the ESP32. Need to deal with that. (config.h, and 2 in configReset.cpp).

I didn't see that anyone fixed this, so it's still an issue, I think.

ba58smith, 2018-12-27:
Update: I think that some esp32 boards define LED_BUILTIN and some don't. So, looks like we need something like this, in config.h, maybe?
```
#ifndef LED_BUILTIN
#define LED_BUILTIN 13 // pin number is specific to your esp32 board
#endif
```

ba58smith, 2018-12-27:
Well, that won't work, at least for my Heltec Wifi Kit 32 with the built-in screen. It has this in one of its board definition files: ` static const uint8_t LED_BUILTIN = 25;`. So I can't `#define LED_BUILTIN x` because I guess #ifndef doesn't know that LED_BUILTIN has already been used as a constant variable. Is there another way for me to detect that LED_BUILTIN has already been used, but not with a #define?

mxtommy, 2019-01-01:
merged #69 :) That works fine :)

Thanks!

closed 2019-01-01 by mxtommy