mxtommy/SigkSens #70: The 3 "Local" sensors show a value when rebooted

ba58smith, 2018-12-31:
It appears that the three Local sensors (systemHz, freeMem, and uptime) store a value at some point and display that value when the MCU is rebooted. Other sensors show a value of "null" after a reboot. Not a big deal, but potentially confusing, especially the "uptime" value, which you would expect to show a very low number, if it shows a number. (Mine always shows 207.46 seconds, see below.)

  "sensors": [
    {
      "address": "Local",
      "type": 0,
      "attrs": [
        {
          "name": "systemHz",
          "signalKPath": "sensors.WemosD1Lite.systemHz",
          "value": "45223.00"
        },
        {
          "name": "freeMem",
          "signalKPath": "sensors.WemosD1Lite.freeMem",
          "value": "15512"
        },
        {
          "name": "uptime",
          "signalKPath": "sensors.WemosD1Lite.uptime",
          "value": "207.46"
        }
      ]
    },

mxtommy, 2019-01-01:
It's just an artifact from the way the config is generated/saved :) While confusing it doesn't harm anything :) If we go with the new Key/Value store, this will go away :) Will leave issue open until it's gone :)

mxtommy, 2019-05-22:
Not an issue in new dev version :)

closed 2019-05-22 by mxtommy