mxtommy/SigkSens #72: I can`t set sensors anymore with latest version

hwater, 2019-02-19:
_With my old ESP8255 the Setting of the Inputs is no Problem, the return "true" after the Websetting.
But with the actual Version, the Settings return only "false"_

_Maybe i do some simple Error, please have an Look:
the following return TRUE and the Settings are written:_

http://10.10.10.53/setSignalKToken?token=xxxxxxxhashxxxxx
http://10.10.10.53/setNewHostname?hostname=Achtern
http://10.10.10.53/setSignalKPort?port=3000
http://10.10.10.53/setSignalKHost?host=liesbeth.local

_but if i try to set any real Port my Browser-result is allways FALSE:_

http://10.10.10.53/setSensorAttr?address=D1&attrName=freq&path=propulsion.AchternWelle.revolutions&scale=1
http://10.10.10.53/setSensorAttr?address=A0&attrName=a0&path=electrical.AchternRuder.voltage

The Wemos D1 modules connect fine to openplotter and the Local Values update also. 

I tried to downgrade to the included libraries, but no change. 

Many Greetings from the Danube, 
Holger Watermann

ba58smith, 2019-02-20:
Holger, I have looked at your "commands" and they look correct. I tried to run them, and I also got a "false" return. However, I don't have either digitalIn nor analogIn enabled in the program that I've loaded onto my microcontroller. Maybe that's why you and I are both getting "false"? Please make sure you have enabled those two sensors. See the "Add a Sensor" page in the wiki for more details.

hwater, 2019-02-20:
Hello Brian, all the Options are enabled, even i2c, but without the gyro chip.
I tried it also with an onewire sensor connected, ist is recognized but the same false result.
i2c chips are recognized in the init serial string - but the same result.

ba58smith, 2019-02-20:
Holger, what branch in GitHub did you start with? The Master branch?

hwater, 2019-02-20:
yes

ba58smith, 2019-02-20:
I think @mxtommy suggests you enable only the sensors you're going to use, to save some memory - but I don't think that's what's causing the "false" return. I wish I could be more help, but I think this might need input from @mxtommy or @mairas. Sorry!

hwater, 2019-02-20:
thx Brian, so you get the same probs. lets wait for final master. i will use the 8266 from last summer for now.

mxtommy, 2019-05-22:
Sorry a bit late, can you past the config (it should print to terminal when booting the ESP) :)

hwater, 2019-12-30:
Hello Tommy
... also a bit late, just doing some installs on annother boat.

The situation ist the same with the latest master version, ther must have been something wrong in **httpd.cpp** at _httppSetSensorAttr_. as it returns allways false. 

For me it seems the Routine does not do the SensorInfo correct. Even the simple set Analog Input refuses to work. /setSensorAttr?address=A0&attrName=a0&path=electrical.Battery.voltage

I help me with copying my old working esp8266 and flash it again on an new one. This works most of the times - but not with some HW-Modules.

Please have an look on the httpd.cpp and its versions (my working one is somewhere from 2017-8)

PS: my config is nothing special, only enabled the Ain, Dout, Onewire and Din Ports plus undef ENABLE_SERIAL_DATA.

Happy New Year
Holger

ba58smith, 2019-12-30:
@hwater , please copy your configuration information here. (Look at this section in the wiki - this is the info I would like for you to post here: https://github.com/mxtommy/SigkSens/wiki/Getting-Started#first-test)

hwater, 2019-12-30:
Okay Brian, here is my Config (LoLin HW with latest master)

15:35:22.618 -> rl d⸮⸮| ⸮l⸮|⸮$⸮c|⸮⸮⸮⸮{⸮c⸮c⸮⸮o'⸮lno⸮⸮⸮bx⸮l;l{lx⸮n⸮⸮d⸮⸮cg⸮|l⸮l⸮c⸮⸮'o⸮ $⸮⸮l`⸮o'l`o;Ǜ⸮'⸮d`{ۓo⸮$`⸮r⸮p⸮l⸮{⸮`⸮⸮g⸮mounted file system
15:35:22.724 -> FS Contents:
15:35:22.758 -> /web/index.html / 15
15:35:22.758 -> /config.json / 536
15:35:22.758 -> *WM: Adding parameter
15:35:22.758 -> *WM: myHostname
15:35:22.758 -> *WM: 
15:35:22.758 -> *WM: AutoConnect
15:35:22.758 -> *WM: Connecting as wifi client...
15:35:22.758 -> *WM: Using last saved values, should be faster
15:35:27.266 -> *WM: Connection result: 
15:35:27.266 -> *WM: 3
15:35:27.266 -> *WM: IP Address:
15:35:27.266 -> *WM: 172.20.10.2
15:35:27.266 -> Connected to Wifi!
15:35:27.266 -> *WM: freeing allocated params!
15:35:27.266 -> reading config file
15:35:27.266 -> opened config file
15:35:27.300 -> Current Configuration:
15:35:27.300 -> {
15:35:27.300 ->   "hostname": "SE",
15:35:27.300 ->   "signalKHost": "eleonore.local",
15:35:27.300 ->   "signalKPort": 80,
15:35:27.300 ->   "signalKPath": "/signalk/v1/stream",
15:35:27.300 ->   "signalKToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6Im9wZW5wbG90dGVyIiwiaWF0IjoxNTc3MzEyNTEwLCJleHAiOjQ3MDE1MTQ5MTB9.Pfd-ymx_jog486IU2x-ddShuuE7ewmTkfT-3j7Umd-M",
15:35:27.300 ->   "sensors": [
15:35:27.300 ->     {
15:35:27.300 ->       "address": "Local",
15:35:27.300 ->       "type": 0,
15:35:27.337 ->       "attrs": [
15:35:27.337 ->         {
15:35:27.337 ->           "name": "systemHz",
15:35:27.337 ->           "signalKPath": "sensors.SE.systemHz",
15:35:27.337 ->           "value": "49090.00"
15:35:27.337 ->         },
15:35:27.337 ->         {
15:35:27.337 ->           "name": "freeMem",
15:35:27.337 ->           "signalKPath": "sensors.SE.freeMem",
15:35:27.337 ->           "value": "29008"
15:35:27.337 ->         },
15:35:27.337 ->         {
15:35:27.337 ->           "name": "uptime",
15:35:27.337 ->           "signalKPath": "sensors.SE.uptime",
15:35:27.337 ->           "value": "507.76"
15:35:27.374 ->         }
15:35:27.374 ->       ]
15:35:27.374 ->     }
15:35:27.374 ->   ]
15:35:27.374 -> }
15:35:27.374 -> mDNS responder started at SE
15:35:27.374 -> SSDP Started
15:35:27.374 -> Websocket client starting!
15:35:27.374 -> starting webserver
15:35:27.374 -> Ready.
15:35:32.622 -> [WSc] Disconnected!
15:35:42.640 -> Websocket client starting!
15:35:48.066 -> [WSc] Disconnected!

hwater, 2019-12-30:
Hello Brian, thanks for responding. I just tried your old Fork and i get the same false Err. 
All my libs are actual - only the json stays at 5.x.

ba58smith, 2019-12-30:
That looks just like mine, so no help there.

Look at your config.h file, for `MAX_SIGNALK_PATH_LEN`. I don't know why that would ever be changed, but make sure it's 150. If it's too short, I don't think SetSensorAttr() would succeed.

mxtommy, 2019-12-30:
Hi!

Have you enabled the sensors at the beginning of config.h?

`//#define ENABLE_DIGITALIN
//#define ENABLE_DIGITALOUT
//#define ENABLE_ANALOGIN
//#define ENABLE_ONEWIRE`

Been a while but it should show the sensors in the config if they are enabled.

Thanks!

hwater, 2019-12-30:
Good Tip, Brian. But even with LEN=250 it did not work. 
Memory is 4bit in my module so the stack should also be well.

But normaly the old working Source shows an List with the enabled IOs and found OneWires or I2C Adresses. This is no more the case here. Maybe we look at the wrong place. 

Can´t belive i am the only one with this porblem...

Holger

hwater, 2019-12-30:
hey, correct. i did not enable read my_config in your fork - so no port was enabled...

try it again now

hwater, 2019-12-30:
Success, for some reason it did not use my_config.h. so i commentet it again and voiala:

mDNS responder started at SE
16:25:18.505 -> SSDP Started
16:25:18.505 -> Websocket client starting!
16:25:18.538 -> 1-Wire Parasite power is: OFF
16:25:18.538 -> 1-Wire Device precision currently: 9 setting to 10 Done!
16:25:18.538 -> Starting 1-Wire scanning timer at: 30000ms
16:25:18.538 -> Scanning 1-Wire Bus
16:25:18.538 -> Scanning for i2c Sensors...
16:25:18.538 -> Setting up Digital Input on pin: D1
16:25:18.538 -> Setting up Digital Input on pin: D2
16:25:18.538 -> Initializing digitalOut pin: 16
16:25:18.538 -> Setting up Digital Output on pin: OUT1
16:25:18.538 -> Initializing digitalOut pin: 15
16:25:18.538 -> Setting up Digital Output on pin: OUT2
16:25:18.575 -> Adding A0 input 
16:25:18.575 -> saving config
16:25:18.575 -> Saving sensor Local
16:25:18.575 -> Saving sensor D1
16:25:18.575 -> Saving sensor D2
16:25:18.575 -> Saving sensor A0
16:25:18.575 -> Saving sensor OUT1
16:25:18.575 -> Saving sensor OUT2
16:25:18.575 -> {

Thanks for your Time - Happy new Year and all... Holger

ba58smith, 2019-12-30:
Ah! I never would have thought of looking in myconfig.h, instead of just config.h. Glad you did, and glad that worked! I will close this Issue.

closed 2019-12-30 by ba58smith