mxtommy/SigkSens #73: FunctionalInterrupt.h

AliceJ21, 2019-04-10:
upon compiling, it continually fails with :

FunctionalInterrupt.cpp:1:33: error: FunctionalInterrupt.h: No such file or directory

 #include <FunctionalInterrupt.h>

I have tried manually putting the .cpp and .h files into the sketch but it still comes up with this error.

ba58smith, 2019-04-10:
AliceJ21, I think I had that problem when I was first trying to compile for an ESP32. It ended up being the wrong version of a library, I think. If you're compiling for an ESP32, be sure to read the detailed instructions I wrote in the Getting Started document in the wiki.

AliceJ21, 2019-04-10:
I have managed to crack it - the esp8266 library for Arduino was not installed! Thanks

closed 2019-04-10 by AliceJ21