mxtommy/SigkSens #74: INA219 Compatibilty

AliceJ21, 2019-04-11:
Is the I2C Sensor INA219 supported yet? I have managed to get the address showing in the Serial Monitor but there is no definition of it?

ba58smith, 2019-04-11:
In the "Add a Sensor" wiki, you see all the types of i2c sensors that are currently supported, and the INA219 isn't in that list. I haven't tried to add a new type of sensor yet, but with several i2c sensors already defined, maybe you can add it yourself by using one of them as a template? Look in SigkSens/SigkSens/src/sensors for examples. Otherwise, @mxtommy or @mairas will need to help.

mxtommy, 2019-05-22:
Unfortunately I do not have one to test with :( If you want help coding it let me know, but not much I can do without the hardware :)

closed 2019-05-22 by mxtommy