mxtommy/SigkSens #75: Exception (28) After Configuring WIFI

dagnum, 2019-05-15:
I had it working perfectly for quite some time.  Suddenly, now I cant figure out how to fix this problem.  I load and flash all contents, Connect to Unconfigured sensor and load the wifi password.  After resetting my ESP8266 it will just keep crashing and crashing over and over again.

Sorry to bother you guys its been happening for months and really would love to get back to developing my ESP8266 with this great code!

[image]

*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.89
Connected to Wifi!
*WM: [3] freeing allocated params! 
*WM: [3] unloading 
reading config file
opened config file
Current Configuration:
{
  "hostname": "",
  "signalKHost": "",
  "signalKPort": 80,
  "signalKPath": "/signalk/v1/stream",
  "signalKToken": "",
  "sensors": []
}
mDNS responder started at 
SSDP Started

Exception (28):
epc1=0x4022fed4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 01a0
3ffffeb0:  00000000 4023ef30 00000028 3fff1280  
3ffffec0:  3ffe89ee 3ffe8eee 00000000 4021664f  
3ffffed0:  0000204a 3fff3e34 3fffff20 40100864  
3ffffee0:  00000008 00000471 3fffff00 00000001  
3ffffef0:  00000000 3fff1280 3fff12f4 402168d8  
3fffff00:  7300006e 3fff3e34 3fff12f4 3fff167c  
3fffff10:  00000000 3fff1280 3fff12f4 4021c46a  
3fffff20:  3fff2904 2d363632 4021bcd4 4022c358  
3fffff30:  3fffdad0 00000000 3fff12f4 4020b9ae  
3fffff40:  3fff3f6c 4022b820 4020b8b4 4020b7b8  
3fffff50:  3fffdad0 00000000 3fff1550 40209231  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefef00 feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe 3fff164c 4020dd1e  
3fffff90:  feefeffe feefeffe feefeffe 4020dd83  
3fffffa0:  feefeffe feefeffe feefeffe 40227b88  
3fffffb0:  feefeffe feefeffe 3ffe8688 401008f9  
<<<stack<<<

ba58smith, 2019-05-19:
This is just a wild guess, but since the exception occurs right after SSDP starts - do you have the correct SSDP library? If you've tried to get SigkSens working with an ESP32, you would have had to change to a diifferent version of that libary - maybe you did that, but didn't change back when you went back to the ESP8266?

ba58smith, 2019-05-19:
Here's another thought: in discovery.cpp, handleDiscovery( ), I see this:
device.modelName((char*)"ESP-32");
  device.modelNumber(1, 0);
  device.friendlyName((char*)"ESP32-WROVER

@mxtommy - is that maybe leftover from when you were trying to get your ESP32 to work? Should it still be in there?

dagnum, 2019-05-19:
Thank you So much!  I will reload that library and I guess all the libraries.  I never tried the ESP32.  Thank you very much for the reply!

dagnum, 2019-05-19:
Same thing...Just reloads and reloads.  I am going to restart everything from a new computer in case there is something I am missing.  Thank you again.  What you guys have done here is truly amazing.  I am working on this to setup all sorts of boat sensors and tie it in with an alerting system.

mxtommy, 2019-05-22:
Hi,

I see that hostname is empty in the config. That's likely why it's crashing. Are you leaving it blank when configuring the wifi? If not maybe something's changed in the handling of the wifi configuration code. Can you capture the serial output when configuring the wifi for the first time?

Thanks!

dagnum, 2019-05-22:
I cant believe my incompetence.  Added hostname and problem solved.  I am just so used to it not being required in other things please forgive me and thank you.  You wouldnt believe how much time i spent troubleshooting this.  Also this is one of my first Arduino Projects and its truly amazing thank you guys so much.  Let the data flow...

[image]

closed 2019-05-22 by dagnum

ba58smith, 2019-05-22:
I will update the wiki to point out any fields that MUST be completed.

B

ba58smith, 2019-05-26:
@dagnum - Now that I'm reading your post again, I remember that I've had some issues lately with the Hostname being saved when I do the wifi config. I've had to use /setNewHostname to explicitly set it for the last two Wemos D1 Mini's I've put into service. @mxtommy, I mentioned this to you, but I never opened an Issue for it. I will do that now.

dagnum, 2020-10-16:
Hey guys, I have a PCB I had designed for detecting DC current.  I would love to share it with you guys!  let me know how i can show it to you!

CONFIDENTIALITY STATEMENT: The information in this e-mail is confidential and may be legally privileged. It is intended for the use only of the addressee(s). Thank you.

ba58smith, 2020-10-17:
This project is not very active, @dagnum . If you want to share a PCB that uses an ESP8266 or ESP32 to read DC current, your best place is probably the "sensors" channel on signalk-dev.slack.com.