mxtommy/SigkSens #77: 8266 Compile Error

ssna, 2019-10-15:
I'm trying to get SigkSens onto a ESP8266 and getting errors compiling.  I had to downgrade ArduinoJson to 5.13.5 but still get compile errors.  I did have this 2866 transmitting nmea0183 UDP data using micropython.  The SigkSens package looks more attractive if I can get it running. 
 New to SignalK and appreciate any hints.

ErrorMsg
======
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, DOUT (compatible), 512K (32K SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
In file included from D:\ssnash\OneDrive\#Projects\libraries\uSSDP-master\uSSDP.cpp:10:0:

D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/include/ip_addr.h:71:34: error: declaration of C function 'uint32 ipaddr_addr(const char*)' conflicts with

 uint32 ipaddr_addr(const char *cp);

                                  ^

In file included from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip_addr.h:43:0,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/include/ipv4_addr.h:38,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/arch/cc.h:93,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/arch.h:48,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/debug.h:40,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwipopts.h:3581,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/opt.h:51,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/init.h:40,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/IPAddress.h:27,

                 from D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,

                 from D:\ssnash\OneDrive\#Projects\libraries\uSSDP-master\uSSDP.h:11,

                 from D:\ssnash\OneDrive\#Projects\libraries\uSSDP-master\uSSDP.cpp:7:

D:\ssnash\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:204:7: error: previous declaration 'u32_t ipaddr_addr(const char*)' here

 u32_t ipaddr_addr(const char *cp);

       ^

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ba58smith, 2019-10-16:
@ssna, did you go through all of the steps of the Getting Started page in the Wiki? https://github.com/mxtommy/SigkSens/wiki/Getting-Started

ssna, 2019-10-18:
I did. Had a few issues.  The link to AsyncTCP gives 404 not found.ill try again tomorrow.

ba58smith, 2019-10-19:
@ssna, if you can't get it sorted out, you can email me directly for more timely responses. ba 58 smith at gmail dot com. (no spaces, obviously).

Higs99, 2019-11-13:
Hi - did this get resolved as I am having exactly the same issue?  Any help would be really welcome :)

ba58smith, 2019-11-14:
@Higs99, I don't think so. Email me directly if you want - see my comment above for the email address.

ba58smith, 2019-12-30:
I believe this has been resolved, by using an older version of the uSSDP library. The one that works has been uploaded to /libraries, here:
https://github.com/mxtommy/SigkSens/blob/master/Libraries/uSSDP-master-for-SigkSens.zip

mxtommy, 2020-02-20:
Broken/changed libraries in Arduino will be the death of me... Thanks @ba58smith for providing support! Will close this issue for now!

closed 2020-02-20 by mxtommy