mxtommy/SigkSens #78: Running SigKSens is eating up memory

kdschmidt1, 2019-10-28:
hello,
i installed SigKSens and observed that it is eating up the free memory during running (approx. 576bytes every 10 seconds) leading to a reset every ~450 seconds.
what's wrong??

ba58smith, 2019-10-28:
I've not experienced that. My freemem bounces around a very little bit, but doesn't continually decrease. What sensors do you have connected?

kdschmidt1, 2019-10-28:
currently i'm only using the system sensors (code as is) on a wemos d1 mini (esp8266 with the https://arduino.esp8266.com/stable/package_esp8266com_index.json).
everything else is working fine.
the program is ending with an Exception (29) at approx 3kB memory.

ba58smith, 2019-10-28:
Did you follow the detailed instructions for getting it working on the Getting Started page of the wiki? There are references to a lot of specific libraries there - if you don't have all the correct versions, that could be causing the memory leak. Or, possibly, some of those specific libraries have released new versions since the last time I compiled and uploaded to my D1 mini, and maybe a memory leak has been introduced?

kdschmidt1, 2019-10-28:
I have observed the incompatibilities already (I. e. Jason lib) and so used only the libs provided in the sigksens github source. So should be all OK. But for sur there might be differences in the specific 2866 libs, so next, I will try a esp32 with the espressiv source (already ordered). I really don't know what to do else.

ba58smith, 2019-10-28:
The ESP32 is not very thoroughly tested. I got it to work, but only with the system "sensors", not any actual sensors. If @mxtommy doesn't chime in here, you may want to look at another very similar project that's being actively developed: https://github.com/SignalK/SensESP

kdschmidt1, 2019-10-29:
hello ba85smith,
would it be possible to tell me, which version of the 2866 core you have used for the compilation (i'm using 2.5.2). i would like to try a switchback to this core to solve my problem.

ba58smith, 2019-10-29:
@kdschmidt1, I see this in the first few lines when I compile it: `Using core 'esp8266' from platform in folder: /home/butch/.arduino15/packages/esp8266/hardware/esp8266/2.4.2`, so I assume it's 2.4.2. I haven't updated any of the libraries used by SigkSens in many months. Good luck!

kdschmidt1, 2019-10-30:
ok, was also not working (same result). i gave up and changed over to sensesp, that is working properly,
thanks, klaus

closed 2019-10-30 by kdschmidt1