mxtommy/SigkSens #79: BME280 not showing up

thecivvie, 2020-01-10:
I attached a BMP280 to a Wemos D1 Mini. uploaded the ino and in the serial port it shows an i2c found at 76. but no data is sent to the signalK server. I have the local sensors and they show up ok. I tried to set attribs but it comes up with false

http://10.10.10.127/setSensorAttr?address=0x76&attrName=tempK&path=environment.cabin.temp

Any ideas what I maybe doing wrong

ba58smith, 2020-01-10:
Did you go through the Getting Started page on the wiki? Every step?

thecivvie, 2020-01-11:
Followed every step. I had 2 issues. I had to downgrade ArduinoJson, as per a previous issue and  AsyncTCP, there is no idf-update page. I enabled i2c in the config and the enable_bmp280. Mine shows up as 0x76 not 0x77. I have used it in another sketch and I know it works.

thecivvie, 2020-01-11:
I plugeed the BME into my Pi and it works fine there. Other than openplotter doesn't recognise the temperature, well doesn't read it

ba58smith, 2020-01-12:
@thecivvie - When you say your BMP280 "shows up as 0x76 not 0x77" - where are you seeing that? I mean, how do you know yours is at address 0x76? 

Also, please explain this more thoroughly: "I checked the add a sensor page and I noticed that there is no type in the found line just says 76." I don't know what you mean by "no type in the found line".

hwater, 2020-01-12:
i had the same issure with my 0x76 bme280, but after setting its addresspin to 0x77 on the breadboard  it showed magically up on next reset.

hwater, 2020-02-17:
I put together an additional Sensorkit for this "BME280" I2C Chip with sensors for humidity, altitude, temp and pressure. What is the best way to upload the files to here? 
For the BME280 you also need to download another library "Adafruit_BME280".

This works really well, better Huminity Sensor as these DHTxx Chips.

ba58smith, 2020-02-17:
@hwater - SigkSens isn't designed as a generic "sensor container" that would allow you to just upload some files and have it start working. Each of the sensors it works with was coded specifically for the sensor, with the exception of analog sensors that can be read with AnalogRead() and digital sensors that can be read with DigitalRead().
In order to work with your BME280, you would need to start with a copy of the BMP280 folder and files, and then modify the code to add the additional sensor attributes (humidity and altitude - since the BMP280 has only temp and pressure). Then you'd have to search all of the code for references to the BMP280, and duplicate that code for the BME280, in config.h and SigkSens.ino, for example.
If you really want to use this sensor, I suggest you have a look at SensESP as an alternative to SigkSens (https://github.com/SignalK/SensESP). It would be much easier to add this sensor to that project, and that project is in much more active development than SigkSens (no offense, @mxtommy).

hwater, 2020-02-17:
Yes Brian, you are right with this Container Thing, but i managed to do all the integration for the new sensor.
Next Time i should do it with diff Files…
… but i relearned some programming by doing this extension.

inside are all added/changed files.

Ahoi - holger

ba58smith, 2020-02-17:
@hwater - Oh, now I understand. You already did all those changes! Cool! You say "inside are all added/changed files" - but I don't see them. Can you please make a .zip file of them (or, if there are not too many of them, just attach them all individually), and attach them to your reply to this? Please do it from within GitHub, NOT from the email where you may be reading this message. Then I will try to make a Pull Request, to add them to the project. Thank you. (You can drag and drop a file right into the comment box.)

hwater, 2020-02-17:
[SigkSens_additions_BME280.zip](https://github.com/mxtommy/SigkSens/files/4213788/SigkSens_additions_BME280.zip)

Hi Brian - i hope you will recieve the zip

ba58smith, 2020-02-17:
@hwater - I received it, and have successfully compiled it. I don't have a BME280, so I can't test it, but since you say it works, I will create a Pull Request. (I have reviewed every line of the changes you made - all looks good.)

ba58smith, 2020-02-17:
@hwater - Tell me about this line in bme280.cpp: `#define SEALEVELPRESSURE_HPA (1022.75) // (1013.25)`. (I moved it to config.h.) Isn't this value going to be different every day, even multiple times during the day? (This is why I don't think you can really use the altitude output from this sensor, but maybe I'm wrong.)

hwater, 2020-02-17:
I got it From the original BOSCH Driver, i think this ist the normalized Pressure at N.N. Level, or NormalNull or at SeaLevel. Hopefully the Oceans will not rise so fast. I live at 250m, so i corrected a bit, the orig. Value is in ().

ba58smith, 2020-02-17:
@hwater That value changes all the time - all day, every day. So I don't think this sensor is going to be useful for determining altitude. 

The Adafruit guide says "However, you can only really do a good accurate job of calculating altitude if you know the hPa pressure at sea level for your location and day! The sensor is quite precise but if you do not have the data updated for the current day then it can be difficult to get more accurate than 10 meters." 

I changed the value to "standard" (1013.25), and added a comment about the unreliable nature of the Altitude output from this sensor.

hwater, 2020-02-17:
@ba58smith - this is correct, hopefully my ship will continue to swim, and this value will never bo negative ,-)

ba58smith, 2020-02-17:
Pull Request created: https://github.com/mxtommy/SigkSens/pull/84. Waiting for @mxtommy to review and approve.

mxtommy, 2020-02-20:
Approved!
Thanks to both of you!

closed 2020-02-20 by mxtommy