mxtommy/SigkSens #81: Getting Started update: AsyncTCP idf-update tree not available

bhavers, 2020-01-12:
Hi, the Getting Started guide says that you should download the "idf-update" branch AyncTCP.
In the latest version that branch does not exist anymore, but the program seems to compile fine with the latest version.

ba58smith, 2020-01-13:
Next time I'm going to make a change, I'll try the latest version. Thanks for the heads up.

andyrbarrow, 2020-01-21:
Mine fails on PlatformIO with "fatal: Remote branch idf-update not found in upstream origin". Were you compiling on Arduino IDE or PlatformIO?

ba58smith, 2020-01-21:
@andyrbarrow, did you try it with the latest version of AsyncTCP? @bhavers says (above) that it worked for him. I've not tested it yet. I use the Arduino IDE, but I'm sure you can tell PlatformIO to use the latest version.

bhavers, 2020-01-21:
i didn't find a idf-update branch, so used the master branch (on Arduino). Use this link https://github.com/me-no-dev/AsyncTCP/tree/master

ba58smith, 2020-01-21:
I just downloaded the current AsyncTCP Master and it compiled, and it worked. So I changed the Getting Started Wiki to point to that, and I'm closing this Issue.

closed 2020-01-21 by ba58smith