mxtommy/SigkSens #82: OneWire to Celsius

hwater, 2020-01-12:
The "TempK" attribute works well, but setting to "TempC" allways gives an "false" return.
This would be no big Problem, but the "offset" attribute fails also :-(

hwater, 2020-01-12:
for now i use the "conversion" plugin in SignalK-Server to do the math k to c to solve this.

ba58smith, 2020-01-13:
TempK is always what SigkSens should output, because that's what Signal K expects - all temps are to be in Kelvin. It's up to the consumer of the data (such as Instrument Panel) to convert to Celsius or Fahrenheit. See the selection in Instrument Panel?
[image]

closed 2020-01-13 by ba58smith

hwater, 2020-01-13:
Hello Brian, 
thanks for Answering.

Yes, you are correct, but i found the prepared conversion to TempC in the source of onewire.c...

My Onewire Temperature Sensors from SigkSens in Instruments Panel are not configurable, maybe
there ist an special Flag to set to get them configurable. Some Temperature Sources (aka. Raspi.temp) are
settable as in your Pict. 

For now i use the Calibration plugin in SKServer to do the Maths.

regards
Holger

ba58smith, 2020-01-13:
The gauges in Instrument Panel are configurable for certain things (like the temperature scale) only if they're recognized as that kind of data (temperature). Look at the Signal K spec (http://signalk.org/specification/1.0.4/doc/vesselsBranch.html) and use a path that's a temperature, then you'll be able to select Celsius in the gauge in Instrument Panel.