mxtommy/SigkSens #91: Fatal crash and reset if LOLIN D1 mini powers on WITH 6kHz frequency already present on digital input D1 or D2

jakobwal, 2021-04-30:
Fatal crash and reset if LOLIN D1 mini powers on WITH 6kHz frequency already present on digital input D1 or D2. If signal is removed the device boots as normal with no errors. The signal can then be reapplied without problem and frequency measurement is correct. Here is printout of boot sequence with error:

`mounted file system
FS Contents:
/web/index.html / 15
/config.json / 1123
*wm:[2] Added Parameter: myHostname
*wm:[1] AutoConnect 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: Krucknet
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS 
*wm:[1] STA IP Address: 192.168.1.23
Connected to Wifi!
reading config file
opened config file
Current Configuration:
{
  "hostname": "EngineSensor1",
  "signalKHost": "",
  "signalKPort": 80,
  "signalKPath": "/signalk/v1/stream",
  "signalKToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6InBpIiwiaWF0IjoxNjE5NjQ1NzU1LCJleHAiOjQ3NDM4NDgxNTV9.0jomtYHcCMqWhYbdh6Mqyzabl3voy68Om5sexW39NpA",
  "sensors": [
    {
      "address": "Local",
      "type": 0,
      "attrs": [
        {
          "name": "systemHz",
          "signalKPath": "electrical.EngineSensor1.systemHz",
          "value": "55235.00"
        },
        {
          "name": "freeMem",
          "signalKPath": "electrical.EngineSensor1.freeMem",
          "value": "28032"
        },
        {
          "name": "uptime",
          "signalKPath": "electrical.EngineSensor1.uptime",
          "value": "82515.22"
        }
      ]
    },
    {
      "address": "D1",
      "type": 1,
      "attrs": [
        {
          "name": "state",
          "signalKPath": "propulsion.3GM30F.state",
          "offset": 0,
          "scale": 1,
          "value": "true"
        },
        {
          "name": "freq",
          "signalKPath": "",
          "offset": 0,
          "scale": 1,
          "value": "null"
        },
        {
          "name": "count",
          "signalKPath": "",
          "offset": 0,
          "scale": 1,
          "value": "null"
        }
      ]
    },
    {
      "address": "D2",
      "type": 1,
      "attrs": [
        {
          "name": "state",
          "signalKPath": "",
          "offset": 0,
          "scale": 1,
          "value": "null"
        },
        {
          "name": "freq",
          "signalKPath": "propulsion.3GM30F.freq",
          "offset": 0,
          "scale": 1,
          "value": "0.00"
        },
        {
          "name": "count",
          "signalKPath": "propulsion.3GM30F.count",
          "offset": 0,
          "scale": 1,
          "value": "null"
        }
      ]
    }
  ]
}
mDNS responder started at EngineSensor1
SSDP Started
Found server with IP/Port: 192.168.1.13:3000
Websocket client starting!
starting webserver

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x402323a4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9b0 end: 3fffffc0 offset: 0190
3ffffb40:  40101eed 00000000 00000002 40101abc  
3ffffb50:  00000005 00000000 00000020 40100530  
3ffffb60:  00000002 400042db 4026d0eb 00000001  
3ffffb70:  40103c5f 00040000 00000000 00040000  
3ffffb80:  53000000 40103c5c 00040000 00000020  
3ffffb90:  ffffffff 00000004 00000000 40100811  
3ffffba0:  c0037015 00000001 40104cb5 3ffeebf0  
3ffffbb0:  3ffeebc0 2c9f0300 4000050c 00000022  
3ffffbc0:  3fffc200 40100754 3fffc258 4000050c  
3ffffbd0:  400043a3 00000030 00000019 ffffffff  
3ffffbe0:  60000200 00000002 ffffffff 80000000  
3ffffbf0:  20000000 3fff2bd0 80000000 20208020  
3ffffc00:  00000000 3fff2970 00000414 3fff2bd4  
3ffffc10:  000000e0 00208020 00208100 00000030  
3ffffc20:  40104a75 00080000 00000000 00000020  
3ffffc30:  40100829 00000004 00000000 40100811  
3ffffc40:  c0037015 00000000 00001000 00000000  
3ffffc50:  3ffffdb0 2c9f0300 4000050c 00000022  
3ffffc60:  3fffc200 40100754 3fffc258 4000050c  
3ffffc70:  4000df6a 00000030 0000001a ffffffff  
3ffffc80:  4022cd0a 3fff275c 3fff2ab8 00000100  
3ffffc90:  3fff277c ffffffff ffffffff 3fff2b98  
3ffffca0:  00000000 3fff2970 00000413 3fff22dc  
3ffffcb0:  00206000 3fff295c 3fff2a84 00000030  
3ffffcc0:  00000002 400042db 00008001 00000002  
3ffffcd0:  40004b31 3fff2bac 00000100 00208000  
3ffffce0:  401061cc 0000001d 00000000 00000001  
3ffffcf0:  40100581 3fff2bac 00208000 00208100  
3ffffd00:  00008000 3ffffdf3 00208000 402227d1  
3ffffd10:  0000002e 3ffffd50 40236781 40228a48  
3ffffd20:  3ffffe50 00000006 3ffffe50 4023ac02  
3ffffd30:  3fff2bac 00000100 3ffffd60 00008001  
3ffffd40:  3ffe8d39 00005e00 3fff22dc 3fff2b98  
3ffffd50:  3fff295c 00208000 3fff22dc 402271bc  
3ffffd60:  6769666e 6f736a2e 0000006e 4022ccd6  
3ffffd70:  00000100 3fff275c 00000000 00000100  
3ffffd80:  00000060 0000001e 00000000 40100f50  
3ffffd90:  40103c5f 00040000 00000000 00000100  
3ffffda0:  3fff22dc 00008000 00000004 4022a570  
3ffffdb0:  00000000 00000080 00000000 00001e82  
3ffffdc0:  00000000 00001e81 40229f0c 00000000  
3ffffdd0:  3fff275c 00000000 00000020 00000100  
3ffffde0:  0000001f 00000000 00000000 4000050c  
3ffffdf0:  40203d17 00000030 00000018 3fff0f08  
3ffffe00:  3ffffe60 3ffffe90 3fff22dc 4022bfaa  
3ffffe10:  3ffe8d39 00000000 3ffffe24 3ffffe20  
3ffffe20:  0000001e 3fff0003 3fff4c0c 3fff1134  
3ffffe30:  3ffe8d39 3fffdad0 3fff0f08 00000030  
3ffffe40:  40207359 3fffff30 00000000 3fff0f08  
3ffffe50:  3fffdad0 3ffffe90 3fff22dc 40229919  
3ffffe60:  00000012 40287b2c 00000000 40223221  
3ffffe70:  3ffe8d39 676e6974 62657720 76726573  
3ffffe80:  00007265 3ffe8d39 3fff22d4 402276ee  
3ffffe90:  00003498 3fff4c0c 4023224c 3fff4640  
3ffffea0:  00000000 3fff45e4 00000020 40100f1b  
3ffffeb0:  3ffe8af8 00000000 0a0d0a0d 402280bd  
3ffffec0:  3ffe8d39 3fff465c 3fff0390 40208568  
3ffffed0:  000003e8 000003e8 3fff1134 40233261  
3ffffee0:  40222c30 3fff0dbc 3ffe8af6 40203d1e  
3ffffef0:  40222c30 3fff0dbc 3ffe8af6 40211919  
3fffff00:  3fffdad0 00000012 3fff0dbc 402232b0  
3fffff10:  3fffdad0 00000000 3fff0dbc 402232d4  
3fffff20:  3fffff70 00000000 3fff0dbc 40203de5  
3fffff30:  40206f9c 0000006c 40232260 4023224c  
3fffff40:  40204fa8 4023224c 40232260 3fff0f08  
3fffff50:  3fffdad0 00000000 3fff0dbc 40202aef  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefef00 feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe 3fff0ec8 401000e1  
3fffff90:  feefeffe feefeffe feefeffe 40208368  
3fffffa0:  feefeffe feefeffe feefeffe 40225af0  
3fffffb0:  feefeffe feefeffe 3ffe86b8 401011d5  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld`

jakobwal, 2021-04-30:
I tested now also with 50Hz. That gave me 2 crash+reset similar to above but then boot was successful. If the frequency of the input signal is high, it never manages to boot. This looks like some sort of timing/race condition to me, but I could be wrong.

ba58smith, 2021-04-30:
No one is working on SigkSens. If you want to gather sensor data and send it to Signal K, please look at https://github.com/SignalK/SensESP.